wichers/esphome-comfoair #1: Compile error

verstrepeng, 2020-09-05:
```
INFO Reading configuration /config/esphome/esp_comfoair.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp_comfoair
Processing esp_comfoair (board: lolin32; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 49b5f52 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/esp_comfoair/.pioenvs/esp_comfoair/src/main.cpp.o
Archiving /data/esp_comfoair/.pioenvs/esp_comfoair/lib217/libAsyncTCP-esphome.a
In file included from src/main.cpp:16:0:
src/comfoair.h:5:42: fatal error: esphome/components/uart/uart.h: No such file or directory
compilation terminated.
Compiling /data/esp_comfoair/.pioenvs/esp_comfoair/libb2b/WiFi/ETH.cpp.o
*** [/data/esp_comfoair/.pioenvs/esp_comfoair/src/main.cpp.o] Error 1
========================== [FAILED] Took 1.57 seconds ==========================
```

wichers, 2020-09-16:
That version was not working, please pull latest version. And don't forget to adjust your yaml to my example..

closed 2020-09-16 by wichers

verstrepeng, 2020-09-16:
Many thanks for your support, i pulled the latest version, but still running into issues during compilation, altough a lot further...
`Compiling /data/esp_comfoair/.pioenvs/esp_comfoair/src/main.cpp.o
In file included from src/main.cpp:20:0:
src/comfoair.h: In member function 'virtual esphome::climate::ClimateTraits esphome::comfoair::ComfoAirComponent::traits()':
src/comfoair.h:136:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_dry_mode'
     traits.set_supports_dry_mode(false);
            ^
src/comfoair.h:137:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_only_mode'
     traits.set_supports_fan_only_mode(true);
            ^
src/comfoair.h:144:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_mode_auto'
     traits.set_supports_fan_mode_auto(true);
            ^
src/comfoair.h:145:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_mode_high'
     traits.set_supports_fan_mode_high(true);
            ^
src/comfoair.h:146:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_mode_low'
     traits.set_supports_fan_mode_low(true);
            ^
src/comfoair.h:147:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_mode_medium'
     traits.set_supports_fan_mode_medium(true);
            ^
src/comfoair.h:148:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_mode_off'
     traits.set_supports_fan_mode_off(true);
            ^
src/comfoair.h:149:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_swing_mode_off'
     traits.set_supports_swing_mode_off(false);
            ^
src/comfoair.h:150:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_swing_mode_both'
     traits.set_supports_swing_mode_both(false);
            ^
src/comfoair.h:151:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_swing_mode_vertical'
     traits.set_supports_swing_mode_vertical(false);
            ^
src/comfoair.h:152:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_swing_mode_horizontal'
     traits.set_supports_swing_mode_horizontal(false);
            ^
src/comfoair.h: In member function 'virtual void esphome::comfoair::ComfoAirComponent::control(const esphome::climate::ClimateCall&)':
src/comfoair.h:158:14: error: 'const class esphome::climate::ClimateCall' has no member named 'get_fan_mode'
     if (call.get_fan_mode().has_value()) {
              ^
src/comfoair.h:161:13: error: 'class esphome::comfoair::ComfoAirComponent' has no member named 'fan_mode'
       this->fan_mode = *call.get_fan_mode();
             ^
src/comfoair.h:161:30: error: 'const class esphome::climate::ClimateCall' has no member named 'get_fan_mode'
       this->fan_mode = *call.get_fan_mode();
                              ^
src/comfoair.h:162:21: error: 'class esphome::comfoair::ComfoAirComponent' has no member named 'fan_mode'
       switch (this->fan_mode) {
                     ^
src/comfoair.h:163:14: error: 'CLIMATE_FAN_HIGH' is not a member of 'esphome::climate'
         case climate::CLIMATE_FAN_HIGH:
              ^
src/comfoair.h:166:14: error: 'CLIMATE_FAN_MEDIUM' is not a member of 'esphome::climate'
         case climate::CLIMATE_FAN_MEDIUM:
              ^
src/comfoair.h:169:14: error: 'CLIMATE_FAN_LOW' is not a member of 'esphome::climate'
         case climate::CLIMATE_FAN_LOW:
              ^
src/comfoair.h:172:14: error: 'CLIMATE_FAN_OFF' is not a member of 'esphome::climate'
         case climate::CLIMATE_FAN_OFF:
              ^
src/comfoair.h:175:14: error: 'CLIMATE_FAN_AUTO' is not a member of 'esphome::climate'
         case climate::CLIMATE_FAN_AUTO:
              ^
src/comfoair.h:178:14: error: 'CLIMATE_FAN_ON' is not a member of 'esphome::climate'
         case climate::CLIMATE_FAN_ON:
              ^
src/comfoair.h:179:14: error: 'CLIMATE_FAN_MIDDLE' is not a member of 'esphome::climate'
         case climate::CLIMATE_FAN_MIDDLE:
[... 26 lines cut ...]
src/comfoair.h:472:26: error: 'CLIMATE_MODE_FAN_ONLY' is not a member of 'esphome::climate'
             this->mode = climate::CLIMATE_MODE_FAN_ONLY;
                          ^
src/comfoair.h:475:19: error: 'class esphome::comfoair::ComfoAirComponent' has no member named 'fan_mode'
             this->fan_mode = climate::CLIMATE_FAN_MEDIUM;
                   ^
src/comfoair.h:475:30: error: 'CLIMATE_FAN_MEDIUM' is not a member of 'esphome::climate'
             this->fan_mode = climate::CLIMATE_FAN_MEDIUM;
                              ^
src/comfoair.h:476:26: error: 'CLIMATE_MODE_FAN_ONLY' is not a member of 'esphome::climate'
             this->mode = climate::CLIMATE_MODE_FAN_ONLY;
                          ^
src/comfoair.h:479:19: error: 'class esphome::comfoair::ComfoAirComponent' has no member named 'fan_mode'
             this->fan_mode = climate::CLIMATE_FAN_HIGH;
                   ^
src/comfoair.h:479:30: error: 'CLIMATE_FAN_HIGH' is not a member of 'esphome::climate'
             this->fan_mode = climate::CLIMATE_FAN_HIGH;
                              ^
src/comfoair.h:480:26: error: 'CLIMATE_MODE_FAN_ONLY' is not a member of 'esphome::climate'
             this->mode = climate::CLIMATE_MODE_FAN_ONLY;
                          ^
Generating partitions /data/esp_comfoair/.pioenvs/esp_comfoair/partitions.bin
Compiling /data/esp_comfoair/.pioenvs/esp_comfoair/lib217/AsyncTCP-esphome/AsyncTCP.cpp.o
*** [/data/esp_comfoair/.pioenvs/esp_comfoair/src/main.cpp.o] Error 1
========================= [FAILED] Took 17.26 seconds =========================`

verstrepeng, 2020-09-16:
OK, I just updated to ESPhome version 1.15.1, now it compiles correctly !