wichers/esphome-comfoair #2: API issues

verstrepeng, 2020-09-16:
The API connection to HomeAssistant seems to be disconnected all the time, not sure why. I don't have this problem with other ESPhome devices. This causes stability issues.
```[09:56:23][D][comfoair:545]: ComfoAir getting valve status
[09:56:24][D][comfoair:562]: ComfoAir getting sensor data
[09:56:25][D][comfoair:569]: ComfoAir getting ventilation level
[09:56:26][D][comfoair:575]: ComfoAir getting temperatures
[09:56:27][D][comfoair:537]: ComfoAir getting fan status
[09:56:28][D][comfoair:545]: ComfoAir getting valve status
[09:56:29][D][comfoair:562]: ComfoAir getting sensor data
[09:56:30][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:30][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:30][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:30][D][comfoair:569]: ComfoAir getting ventilation level
[09:56:30][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:30][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:31][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:31][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:31][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:31][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:31][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:31][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:31][D][comfoair:575]: ComfoAir getting temperatures
[09:56:31][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:31][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:32][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:32][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:32][D][comfoair:537]: ComfoAir getting fan status
[09:56:32][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:32][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:33][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:33][D][api.connection:604]: Client 'Home Assistant 0.114.4 (192.168.2.40)' connected successfully!
[09:56:33][D][api:067]: Disconnecting Home Assistant 0.114.4 (192.168.2.40)
[09:56:33][D][api:067]: Disconnecting 192.168.2.40
```
In HomeAssistant log following error:
```Logger: homeassistant.components.climate
Source: helpers/entity_platform.py:462
Integration: Klimaat (documentation, issues)
First occurred: 9:39:47 (1868 occurrences)
Last logged: 9:58:05

Platform esphome does not generate unique IDs. ID esp_comfoairclimatecomfoair already exists - ignoring climate.comfoair
```

wichers, 2020-09-16:
I don't have this issue, might be there because the comfoair is not
connected to the rs232 port. Is the comfoair connected?

verstrepeng, 2020-09-16:
ComfoAir is not connected for the moment. I'll try that later.

It is causing stability issues on HomeAssistant as state changes to often... So should somewhere be caught in the code, else if your comfoair has an issue, it kills your HA slowly.

wichers, 2020-09-16:
Agree, I'll see what I can do..

closed 2022-05-03 by verstrepeng