wichers/esphome-comfoair #3: [Question] Compability with StorkAir G91-350

KM192, 2020-09-24:
I have StorkAir G91-350 from 2009 year:
- is it compatible with this version
- how to connect esp (e.g. D1 Wemos version) with 350 interface ?

wichers, 2020-09-24:
I was unable to find anything about the 'magic'? DB9 connector on this unit. The electrical schematic does show an A and B connected to pin 8 & 9 of the DB9 connector. For me the question is what this A & B is exposed to. From the high-res pcb picture I was unable to draw any conclusions.. At the very least we know the Philips P80C552EFA supports UART. If your adventurous and have a multimeter you could check if pin 24 & pin 25 (easy to find, next to the marked pin 26) of the Philips PLCC chip are directly or indirectly (especially check the two transistors and the 10k resistors near A & B) connected to A & B port.

[image]

pki791, 2020-12-27:
Hi.
A & B are the communiaction pins for RS485. The protocoll seems slightly different, seems it is like here https://matsab.de/de/heimautomatisierung/comfoair

I have done a test and seems it is willing to work. Log below. Is anyone willing to help and add this protocol too?

I have build a small converter TTL to RS485 with MAX485 and NE555, not complicated, auto RX/TX switching.

SEND
07 F0 00 85 00 32 07 0F

GOT
07 F3
07 F0 00 84 0A 00 12 00 30 4E 31 86 BE 01 00 41 07 0F
07 F0 00 84 0A 00 12 00 30 4E 31 86 BE 01 00 41 07 0F
07 F0 00 84 0A 00 12 00 30 4E 31 86 BE 01 00 41 07 0F
07 F0 00 84 0A 00 12 00 30 4E 31 86 BE 01 00 41 07 0F
07 F0 00 84 0A 00 12 00 30 4E 31 86 BE 01 00 41 07 0F

pki791, 2020-12-27:
Interface i used to connect ESP8266 to RS485
[image]

KM192, 2021-02-07:
Hi - I opened my CA 350 and it has RS232 port on the board:
[image]
Do you think and RS232 <> USB Conector to ESP32 with ESPHome shoud work in this case ?

wichers, 2024-04-03:
Closing this issue as the [protocol](https://matsab.de/images/comfoair/Protokoll_CA500_Avignon.pdf) is completely different. This requires a new implementation.

closed 2024-04-03 by wichers