wichers/esphome-comfoair #4: Compile issue

geelenbert, 2020-12-02:
I can't compile the code using the latest ESPHome version:

`INFO Reading configuration /config/esphome/WHR930.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/whr930
Processing whr930 (board: d1_mini; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/whr930/.pioenvs/whr930/src/main.cpp.o
Compiling /data/whr930/.pioenvs/whr930/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling /data/whr930/.pioenvs/whr930/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
In file included from src/main.cpp:16:0:
src/comfoair.h:753:13: error: 'sensor' does not name a type
             sensor::Sensor* fan_supply_air_percentage;
             ^
src/comfoair.h:754:13: error: 'sensor' does not name a type
             sensor::Sensor* fan_exhaust_air_percentage;
             ^
src/comfoair.h:755:13: error: 'sensor' does not name a type
             sensor::Sensor* fan_speed_supply;
             ^
src/comfoair.h:756:13: error: 'sensor' does not name a type
             sensor::Sensor* fan_speed_exhaust;
             ^
src/comfoair.h:757:13: error: 'binary_sensor' does not name a type
             binary_sensor::BinarySensor* is_bypass_valve_open;
             ^
src/comfoair.h:758:13: error: 'binary_sensor' does not name a type
             binary_sensor::BinarySensor* is_preheating;
             ^
src/comfoair.h:759:13: error: 'sensor' does not name a type
             sensor::Sensor* outside_air_temperature;
             ^
src/comfoair.h:760:13: error: 'sensor' does not name a type
             sensor::Sensor* supply_air_temperature;
             ^
src/comfoair.h:761:13: error: 'sensor' does not name a type
             sensor::Sensor* return_air_temperature;
             ^
src/comfoair.h:762:13: error: 'sensor' does not name a type
             sensor::Sensor* exhaust_air_temperature;
             ^
src/comfoair.h:763:13: error: 'sensor' does not name a type
             sensor::Sensor* enthalpy_temperature;
             ^
src/comfoair.h:764:13: error: 'sensor' does not name a type
             sensor::Sensor* ewt_temperature;
             ^
src/comfoair.h:765:13: error: 'sensor' does not name a type
             sensor::Sensor* reheating_temperature;
             ^
[... 488 lines cut ...]
    ca->fan_exhaust_air_percentage = new Sensor("Fan Exhaust Air");
    App.register_sensor(ca->fan_exhaust_air_percentage);
    ca->fan_speed_supply = new Sensor("Fan Speed Supply");
    App.register_sensor(ca->fan_speed_supply);
    ca->fan_speed_exhaust = new Sensor("Fan Speed Exhaust");
    App.register_sensor(ca->fan_speed_exhaust);
    ca->is_bypass_valve_open = new BinarySensor("Bypass Valve");
    App.register_binary_sensor(ca->is_bypass_valve_open);
    ca->is_preheating = new BinarySensor("Preheating");
    App.register_binary_sensor(ca->is_preheating);
    ca->is_supply_fan_active = new BinarySensor("Supply Fan");
    App.register_binary_sensor(ca->is_supply_fan_active);
    ca->is_filter_full = new BinarySensor("Filter Full");
    App.register_binary_sensor(ca->is_filter_full);
    ca->is_summer_mode = new BinarySensor("Summer Mode");
    App.register_binary_sensor(ca->is_summer_mode);
    ca->return_air_level = new Sensor("Return Air Level");
    App.register_sensor(ca->return_air_level);
    ca->supply_air_level = new Sensor("Supply Air Level");
    App.register_sensor(ca->supply_air_level);
    App.register_climate(ca);
    return {ca};
    
  climates:
    - name: "WTW"`

wichers, 2020-12-02:
You're missing the following two lines in your yaml file:

```
sensor:
binary_sensor:

```

closed 2020-12-02 by wichers

geelenbert, 2020-12-02:
Missed it, Works perfect thanks!