wichers/esphome-comfoair #6: Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

erix, 2021-01-05:
I'm getting the following panic when running the software:

[20:12:49][D][comfoair:476]: Level 03
[20:12:49][D][sensor:092]: 'Return Air Level': Sending state 50.00000  with 0 decimals of accuracy
[20:12:49][D][sensor:092]: 'Supply Air Level': Sending state 50.00000  with 0 decimals of accuracy
[20:12:49][D][climate:262]: 'Comfoair 350' - Sending state:
[20:12:49][D][climate:265]:   Mode: FAN_ONLY
[20:12:49][D][climate:270]:   Fan Mode: MEDIUM
[20:12:49][D][climate:282]:   Target Temperature: -169473953031737902729750710990328037376.00°C
[20:12:49][D][binary_sensor:036]: 'Supply Fan': Sending state ON
[20:12:49][D][comfoair:617]: getting ventilation level
[20:12:49][D][comfoair:476]: Level 03
[20:12:49][D][sensor:092]: 'Return Air Level': Sending state 50.00000  with 0 decimals of accuracy
[20:12:49][D][sensor:092]: 'Supply Air Level': Sending state 50.00000  with 0 decimals of accuracy
[20:12:49][D][climate:262]: 'Comfoair 350' - Sending state:
[20:12:49][D][climate:265]:   Mode: FAN_ONLY
[20:12:49][D][climate:270]:   Fan Mode: MEDIUM
[20:12:49][D][climate:282]:   Target Temperature: -169473953031737902729750710990328037376.00°C
[20:12:49][D][climate:262]: 'Comfoair 350' - Sending state:
[20:12:49][D][climate:265]:   Mode: FAN_ONLY
[20:12:49][D][climate:270]:   Fan Mode: MEDIUM
[20:12:49][D][climate:282]:   Target Temperature: 20.00°C
[20:12:49][D][sensor:092]: 'Outside Air': Sending state 1.00000  with 0 decimals of accuracy
[20:12:49][D][sensor:092]: 'Supply Air': Sending state 20.00000  with 0 decimals of accuracy
[20:12:49][D][sensor:092]: 'Return Air': Sending state 22.00000  with 0 decimals of accuracy
[20:12:49][D][sensor:092]: 'Exhaust Air': Sending state 6.00000  with 0 decimals of accuracy
[20:12:50]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[20:12:50]Core 1 register dump:
[20:12:50]PC      : 0x400d8c1e  PS      : 0x00060e30  A0      : 0x800de68d  A1      : 0x3ffb1e80
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d comfoair -t idedata
WARNING Decoded 0x400d8c1e: esphome::sensor::Sensor::publish_state(float) at /Users/q326374/dev/github/esphome-comfoair/comfoair/src/esphome/components/sensor/sensor.cpp:85
[20:12:50]A2      : 0xfefefefe  A3      : 0x40a00000  A4      : 0x40158b3c  A5      : 0x00000020
[20:12:50]A6      : 0x00000000  A7      : 0x0000009e  A8      : 0x00000002  A9      : 0x00000002
[20:12:50]A10     : 0x0000000f  A11     : 0x00000003  A12     : 0x00000003  A13     : 0x3ffafe1c
[20:12:50]A14     : 0x0000002a  A15     : 0x3ffb1f70  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d
[20:12:50]EXCVADDR: 0xfefeff3e  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000
[20:12:50]
[20:12:50]Backtrace: 0x400d8c1e:0x3ffb1e80 0x400de68a:0x3ffb1eb0 0x400de8fe:0x3ffb1ed0 0x400de922:0x3ffb1f00 0x40158c89:0x3ffb1f20 0x40158d55:0x3ffb1f40 0x400dbba2:0x3ffb1f60 0x400de92e:0x3ffb1f90 0x400e3fa9:0x3ffb1fb0 0x40088db5:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400d8c1e: esphome::sensor::Sensor::publish_state(float) at /Users/q326374/dev/github/esphome-comfoair/comfoair/src/esphome/components/sensor/sensor.cpp:85
WARNING Decoded 0x400de68a: esphome::comfoair::ComfoAirComponent::parse_data_() at /Users/q326374/dev/github/esphome-comfoair/comfoair/src/main.cpp:206
WARNING Decoded 0x400de8fe: esphome::comfoair::ComfoAirComponent::loop() at /Users/q326374/dev/github/esphome-comfoair/comfoair/src/main.cpp:206
WARNING Decoded 0x400de922: non-virtual thunk to esphome::comfoair::ComfoAirComponent::loop()
WARNING Decoded 0x40158c89: esphome::Component::call_loop() at /Users/q326374/dev/github/esphome-comfoair/comfoair/src/esphome/core/component.cpp:111
WARNING Decoded 0x40158d55: esphome::Component::call() at /Users/q326374/dev/github/esphome-comfoair/comfoair/src/esphome/core/component.cpp:111
WARNING Decoded 0x400dbba2: esphome::Application::loop() at /Users/q326374/dev/github/esphome-comfoair/comfoair/src/esphome/core/application.cpp:72
WARNING Decoded 0x400de92e: loop() at /Users/q326374/dev/github/esphome-comfoair/comfoair/src/main.cpp:206
WARNING Decoded 0x400e3fa9: loopTask(void*) at /Users/q326374/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
WARNING Decoded 0x40088db5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[20:12:51]
[20:12:51]Rebooting...

This is repeating always at the same place and I cannot figure out what is the problem. Can you please help?

wichers, 2021-01-05:
To help you further I would like to see a copy of your yaml, please post it.

erix, 2021-01-05:
```
esphome:
  name: comfoair
  platform: ESP32
  board: esp32dev
  includes:
    - comfoair.h

wifi:
  ssid: '*******'
  password: '*******'

# Disable uart logging
logger:
  #baud_rate: 0
  level: DEBUG

# Enable Home Assistant API
api:

ota:

sensor:
binary_sensor:

uart:
  id: uart_bus
  baud_rate: 9600
  tx_pin: GPIO17
  rx_pin: GPIO16
  # tx_pin: TX
  # rx_pin: RX

climate:
- platform: custom
  lambda: |-
    auto ca = new esphome::comfoair::ComfoAirComponent(id(uart_bus));
    App.register_component(ca);

    ca->outside_air_temperature = new Sensor("Outside Air");
    App.register_sensor(ca->outside_air_temperature);
    ca->supply_air_temperature = new Sensor("Supply Air");
    App.register_sensor(ca->supply_air_temperature);
    ca->return_air_temperature = new Sensor("Return Air");
    App.register_sensor(ca->return_air_temperature);
    ca->exhaust_air_temperature = new Sensor("Exhaust Air");
    App.register_sensor(ca->exhaust_air_temperature);

    ca->fan_supply_air_percentage = new Sensor("Fan Supply Air");
    App.register_sensor(ca->fan_supply_air_percentage);
    ca->fan_exhaust_air_percentage = new Sensor("Fan Exhaust Air");
    App.register_sensor(ca->fan_exhaust_air_percentage);
    ca->fan_speed_supply = new Sensor("Fan Speed Supply");
    App.register_sensor(ca->fan_speed_supply);
    ca->fan_speed_exhaust = new Sensor("Fan Speed Exhaust");
    App.register_sensor(ca->fan_speed_exhaust);

    ca->is_bypass_valve_open = new BinarySensor("Bypass Valve");
    App.register_binary_sensor(ca->is_bypass_valve_open);
    ca->is_preheating = new BinarySensor("Preheating");
    App.register_binary_sensor(ca->is_preheating);
    ca->is_supply_fan_active = new BinarySensor("Supply Fan");
    App.register_binary_sensor(ca->is_supply_fan_active);
    ca->is_filter_full = new BinarySensor("Filter Full");
    App.register_binary_sensor(ca->is_filter_full);
    ca->is_summer_mode = new BinarySensor("Summer Mode");
    App.register_binary_sensor(ca->is_summer_mode);

    ca->return_air_level = new Sensor("Return Air Level");
    App.register_sensor(ca->return_air_level);
    ca->supply_air_level = new Sensor("Supply Air Level");
    App.register_sensor(ca->supply_air_level);

    App.register_climate(ca);
    return {ca};
    
  climates:
    - name: "Comfoair 350"

```

wichers, 2021-01-06:
Haven't done any testing on esp32 platform, hoping it's related to a nullptr dereference.  Could you checkout the latest version and test again? If this is not working I'll add explicit logging to find what this could be. Please report back.

erix, 2021-01-06:
The new patch seems to work, I don't get any crashes anymore. Thank you for your work and support.

wichers, 2021-01-06:
Cheers!, Thank you for the report, so this was indeed a nullptr dereference.

closed 2021-01-06 by wichers