wichers/esphome-comfoair #11: compatible with Comfoair E400 series?

heldopsokken, 2022-01-01:
I am unable to find anything in the documentation is there is any uart connection available for the Zehnder ComfoAir E400.
Do you know where I could find it. Currently I can only control fan speeds via [nRF905-API](https://github.com/eelcohn/nRF905-API). As nice as this is I'd like to see other sensor values which you can read out in via the display as well.

wichers, 2024-04-03:
I'm sorry for this two year late reply. There does appear to be a connector (C3) that is unused and it looks like there are shift level converters attached to these pins. The pinout needs to be traced back (using a multimeter) to the main processor to confirm this. I'll leave this comment open for someone who is willing to investigate further.

[image]

remmob, 2025-07-06:
Does any one already find out if the c3 connector can be used.
I would like to know too

pve84, 2025-08-27:
[image]
The two lines coming from C3 go through a resistor and connect to a HVD72 TI37K which seems to be a RS-485 chip. This could mean that the protocol is Modbus RTU. Sadly I have not been able to probe the lines or check for continuity yet.

pve84, 2026-02-11:
I have a EW11 which I currently use to get data out of my PV Inverter. Not much data these days so i'll try to hook it up and use a general modbus integration like https://github.com/timlaing/modbus_local_gateway to see if I can get some data in HA.

Looking at the screenshots by @remonvandenbrandt  I'll use.
Baud Rate: 19200
Data Bit: 8
Stop Bit: 1
Parity: EVEN

remmob, 2026-02-11:
Don't use more than 1 device at the same time om the Ew-11 they don't seem to like that.
As I have tried with my SHT20 integration.

I can make a HA integration for this, using RTU or With a gateway RTU to TCP/IP

remonvandenbrandt, 2026-02-11:
I created an ESPHome example and a list of all identified registers here
https://github.com/remonvandenbrandt/zehnder-comfoair

CodedCactus, 2026-02-14:
I'm using the 3-way switch and RF. The 3-way switch is located at 0x14A and the RF setpoint is located at 0x149. 

The unit final ventilation setting should be the max of all three setpoints.