wichers/esphome-comfoair #12: Filter reset command does not reset filter hours

adorobis, 2022-01-20:
Hello,
I'm using this method to control the Comfo Air 350 unit and all works great except the function to reset the filter status. It resets it but after an hour it goes back to the filter full status. When digging deeper I've noticed that it does not reset the filter hours counter (available via command xDB) so when it increases by one hour the filter goes to full status again. Any idea what could be wrong or how to fix it?

adorobis, 2022-01-20:
It resets the filter status but after an hour it goes back to filter full status. Looking at the filter hours I can see that it is not getting reset to zero. Not sure if anybody had similar experience - maybe some issue with my unit's controller or something...

adorobis, 2022-01-30:
Btw, thanks to author of openhab binding for comfoair I was able to fix this by sending command that simulates 2sec "Mode" button press. It works ok now for me (it is x37 command with Number 82 in the second byte of data block means pressing the modus button by 2080 milisec. Full command is `07 f0 00 37 07 00 82 00 00 00 00 00 6d 07 0f`).

closed 2022-01-30 by adorobis

override80, 2024-06-04:
Hi @wichers, filter reset in my CA350 has the same issue that @adorobis described. This should work?

```
static const uint8_t CMD_FILTER_RESET = 0x37;
...
void filter_reset(void) {
    uint8_t reset_cmd[6] = {82, 0, 0, 0, 0, 0};
    this->write_command_(CMD_FILTER_RESET, reset_cmd, sizeof(reset_cmd));
} 
```

I've create a lambda for the reset button like this in the end of comfoair.yaml:

```
button:
  - platform: template
    name: "Reset Filter"
    on_press:
      then:
        - lambda: |-
                auto filter_reset = new esphome::comfoair::ComfoAirComponent();
                filter_reset->filter_reset();
```

but probably this is not even the way to go, because the filter hours does not even reset in my case (instead, it reboots the board).