wichers/esphome-comfoair #13: Is rs232-TTL converter needed?

DamianBroj, 2022-01-30:
Hey,
In case of using nodemcu (esp8266 based)- is it needed to use RS232-TTL converter or connecting TX and RX lines directly to nodemcu will be fine?

wichers, 2022-01-31:
No, you need the rs232 converter, ther're not TTL voltages.

DamianBroj, 2022-01-31:
No matter if i am using db9 or rj45 sockets? What is the logic voltage on CA side? For Nodemcu it's if i am not mistaken 3.3v

wichers, 2022-01-31:
Valid RS-232 signals are either in the range of +3 V to +15 V or the range
−3 V to −15 V with respect to common ground. My advice is to use something
with the max3232 chipset and you're safe.

No idea where the db9 is located, at least the rj45 on the CA is known to
be rs232.

DamianBroj, 2022-01-31:
Got it. So from one MAX3232 side it would be TX, RX, 3.3v and GND from Nodemcu, from the other side it would be TX, RX and GND from CA. No need for connecting voltage from CA side, right?

wichers, 2022-01-31:
That's correct. For your reference you could have a look at this schematic:
https://github.com/wichers/esphome-comfoair/blob/master/doc/schematic.png

closed 2022-02-02 by DamianBroj