wichers/esphome-comfoair #14: Readings precision

DamianBroj, 2022-02-02:
What is the level of precision for values coming from CA? I see on some HA screens that it produces decimals, while in my case there's no decimals in readings. Is that something that is configurable?

DamianBroj, 2022-02-06:
Based on discussions with other developer - max temp readings precision is 0.5°C (calculation of CA output is to be divided by 2 and deducted with 20). In current commit all sensors are set to decimal precision = 0. PR from @jjhuff is increasing precision which makes this issue resolved once implemented

wichers, 2022-02-07:
Precision is based on your client side configuration, no changes necessary
in esp code. Although it should support 0.5 precision, I haven't heard of a
unit that does.

closed 2024-03-31 by wichers