wichers/esphome-comfoair #16: Compile error on ESPHome v2022.3.1

DarkWarden85, 2022-04-04:
Hello,

I have successfully been using your esphome-comfoair implementation in conjunction with my comfoair 550 ventilation system for over half a year.
However, upon updating to ESPHome v2022.3.1, compilation fails. Here is the output from the logs. Any way to fix this?

/config/esphome/ventilation-control.yaml: In lambda function:
/config/esphome/ventilation-control.yaml:121:86: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [34])'
  121 |     ca->is_bypass_valve_open = new BinarySensor("Comfoair Ventilation Bypass Valve");
      |                                                                                      ^
In file included from src/esphome/core/application.h:13,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor()'
   28 |   explicit BinarySensor();
      |            ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)'
   26 | class BinarySensor : public EntityBase {
      |       ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [34]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [34]' to 'esphome::binary_sensor::BinarySensor&&'
/config/esphome/ventilation-control.yaml:123:77: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [32])'
  123 |     ca->is_preheating = new BinarySensor("Comfoair Ventilation Preheating");
      |                                                                             ^
In file included from src/esphome/core/application.h:13,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor()'
   28 |   explicit BinarySensor();
      |            ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)'
   26 | class BinarySensor : public EntityBase {
      |       ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [32]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [32]' to 'esphome::binary_sensor::BinarySensor&&'
/config/esphome/ventilation-control.yaml:125:84: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [32])'
  125 |     ca->is_supply_fan_active = new BinarySensor("Comfoair Ventilation Supply Fan");
      |                                                                                    ^
In file included from src/esphome/core/application.h:13,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor()'
   28 |   explicit BinarySensor();
      |            ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)'
   26 | class BinarySensor : public EntityBase {
      |       ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [32]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [32]' to 'esphome::binary_sensor::BinarySensor&&'
/config/esphome/ventilation-control.yaml:127:79: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [33])'
  127 |     ca->is_filter_full = new BinarySensor("Comfoair Ventilation Filter Full");
      |                                                                               ^
In file included from src/esphome/core/application.h:13,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor()'
   28 |   explicit BinarySensor();
      |            ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)'
   26 | class BinarySensor : public EntityBase {
      |       ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [33]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [33]' to 'esphome::binary_sensor::BinarySensor&&'
/config/esphome/ventilation-control.yaml:129:79: error: no matching function for call to 'esphome::binary_sensor::BinarySensor::BinarySensor(const char [33])'
  129 |     ca->is_summer_mode = new BinarySensor("Comfoair Ventilation Summer Mode");
      |                                                                               ^
In file included from src/esphome/core/application.h:13,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor()'
   28 |   explicit BinarySensor();
      |            ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:28:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(const esphome::binary_sensor::BinarySensor&)'
   26 | class BinarySensor : public EntityBase {
      |       ^~~~~~~~~~~~
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [33]' to 'const esphome::binary_sensor::BinarySensor&'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note: candidate: 'esphome::binary_sensor::BinarySensor::BinarySensor(esphome::binary_sensor::BinarySensor&&)'
src/esphome/components/binary_sensor/binary_sensor.h:26:7: note:   no known conversion for argument 1 from 'const char [33]' to 'esphome::binary_sensor::BinarySensor&&'
Compiling /data/ventilation-control/.pioenvs/ventilation-control/lib67b/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling /data/ventilation-control/.pioenvs/ventilation-control/lib67b/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
*** [/data/ventilation-control/.pioenvs/ventilation-control/src/main.cpp.o] Error 1
========================= [FAILED] Took 19.47 seconds =========================

ItsEcholot, 2022-04-10:
I fixed this by adjusting the `.yaml` file:

```
    ca->is_bypass_valve_open = new BinarySensor();
    ca->is_bypass_valve_open->set_name("Bypass Valve");
    App.register_binary_sensor(ca->is_bypass_valve_open);
    ca->is_preheating = new BinarySensor();
    ca->is_preheating->set_name("Preheating");
    App.register_binary_sensor(ca->is_preheating);
    ca->is_supply_fan_active = new BinarySensor();
    ca->is_supply_fan_active->set_name("Supply Fan");
    App.register_binary_sensor(ca->is_supply_fan_active);
    ca->is_filter_full = new BinarySensor();
    ca->is_filter_full->set_name("Filter Full");
    App.register_binary_sensor(ca->is_filter_full);
    ca->is_summer_mode = new BinarySensor();
    ca->is_summer_mode->set_name("Summer Mode");
    App.register_binary_sensor(ca->is_summer_mode);
```

This was caused by the following change in esphome: https://github.com/esphome/esphome/pull/3213

DarkWarden85, 2022-04-11:
Hi @ItsEcholot,

That was it, now everything is working fine again, thank you very much.
I'm closing the issue.

closed 2022-04-11 by DarkWarden85