wichers/esphome-comfoair #17: New mode FOCUS

verstrepeng, 2022-05-03:
I just did an upgrade today and saw a new mode called focus, what is this mode ?

wichers, 2022-05-03:
This commit was for private use, maybe I should not be pushing this to the
main branch, I used "focus" as I needed a function button to handle this
mode. In this mode (at least in my situation) the comfoair will be blowing
more in than out, I personally use this for our fireplace. I still need to
check if I can influence the amount of air being blown in and out. Mine is
set at full speed blowing in and set at the lowest speed blowing out.

To be continued..

verstrepeng, 2022-05-04:
How can I remove the focus setting, as that one is not interesting for me...
Further, great job mate !

wichers, 2024-03-31:
focus has been removed

closed 2024-03-31 by wichers