wichers/esphome-comfoair #18: How to reset the filter full

deam0n, 2022-05-30:
Hi wichers,

how to reset the filter full status?

deam0n, 2022-05-30:
After a closer look, I've added the reset_filter() to the FOCUS FAN MODE:
```
 this->fan_mode = *call.get_fan_mode();
      switch (this->fan_mode.value()) {
        case climate::CLIMATE_FAN_FOCUS:
	  reset_filter();
          break;

        case climate::CLIMATE_FAN_HIGH:
          level = 0x04;
          break;
```

and added the button on the dashboard as follows:
[image]

```
type: vertical-stack
cards:
  - type: custom:simple-thermostat
    entity: climate.comfoair
    hide:
      state: true
      temperature: true
    header:
      name: Ventilation
    control:
      fan:
        'off':
          name: 'Off'
          icon: mdi:fan-minus
        auto:
          name: Auto
          icon: mdi:fan-auto
        low:
          name: Low
          icon: mdi:fan-speed-1
        medium:
          name: Medium
          icon: mdi:fan-speed-2
        high:
          name: High
          icon: mdi:fan-speed-3
        focus:
          name: Reset Filter
          icon: mdi:air-filter
    layout:
      mode:
        headings: false
        names: true
        icons: true
    sensors:
      - entity: sensor.bedroom_sensor_temperature
        name: Bedroom
      - entity: sensor.small_bedroom_motion_temperature
        name: Small Bedroom
      - entity: sensor.bathroom_motion_temperature
        name: Bathroom
      - entity: sensor.bathroom_humidity_sensor_humidity
        name: Bathroom humidity
      - entity: sensor.living_room_sensor_temperature
        name: Living Room
  - type: custom:comfoair-card
    entity: climate.comfoair
```

I got the successfull clear log message! After the first reset filter I was no longer able to reset again (but probably because the filter is now not reporting as full...)

It would be great to have a UI button for this..  Maybe on comfoair.yaml something alike:
```
button:
  - platform: restart
    name: "Comfoair ESP Restart"
```
but for the reset_filter().

therings87, 2023-03-27:
Not sure if its the best way but I solved it by creating a template sensor and a new instance of comfoair:

```
button:
  - platform: template
    name: "Reset Filter"
    on_press:
      then:
        - lambda: |-
                auto reset_filter = new esphome::comfoair::ComfoAirComponent(id(uart_bus));
                reset_filter->reset_filter();
```

May be something to add to the sample config @wichers?

tieskuh, 2023-10-06:
> button:
>   - platform: template
>     name: "Reset Filter"
>     on_press:
>       then:
>         - lambda: |-
>                 auto reset_filter = new esphome::comfoair::ComfoAirComponent(id(uart_bus));
>                 reset_filter->reset_filter();

Hi Therings, 

Can you maybe post your full config? I would like to get a reset button aswell but cannot get it to work with your snippet. I get an error when installing it.

therings87, 2023-10-21:
> Hi Therings,
>
> Can you maybe post your full config? I would like to get a reset button aswell but cannot get it to work with your snippet. I get an error when installing it.

```javascript
esphome:
  name: comfoair
  platform: ESP8266
  board: nodemcuv2
  includes:
    - comfoair.h

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: ""

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:

binary_sensor:

uart:
  - id: uart_bus
    baud_rate: 9600
    tx_pin: TX
    rx_pin: RX

    
sensor:
    
climate:
- platform: custom
  lambda: |-
    auto ca = new esphome::comfoair::ComfoAirComponent(id(uart_bus));
    App.register_component(ca);
    ca->outside_air_temperature = new Sensor("Outside Air");
    App.register_sensor(ca->outside_air_temperature);
    ca->supply_air_temperature = new Sensor("Supply Air");
    App.register_sensor(ca->supply_air_temperature);
    ca->return_air_temperature = new Sensor("Return Air");
    App.register_sensor(ca->return_air_temperature);
    ca->exhaust_air_temperature = new Sensor("Exhaust Air");
    App.register_sensor(ca->exhaust_air_temperature);
    ca->fan_supply_air_percentage = new Sensor("Fan Supply Air");
    App.register_sensor(ca->fan_supply_air_percentage);
    ca->fan_exhaust_air_percentage = new Sensor("Fan Exhaust Air");
    App.register_sensor(ca->fan_exhaust_air_percentage);
    ca->fan_speed_supply = new Sensor("Fan Speed Supply");
    App.register_sensor(ca->fan_speed_supply);
    ca->fan_speed_exhaust = new Sensor("Fan Speed Exhaust");
    App.register_sensor(ca->fan_speed_exhaust);
    ca->is_bypass_valve_open = new BinarySensor();
    ca->is_bypass_valve_open->set_name("Bypass Valve");
    App.register_binary_sensor(ca->is_bypass_valve_open);
    ca->is_preheating = new BinarySensor();
    ca->is_preheating->set_name("Preheating");
    App.register_binary_sensor(ca->is_preheating);
    ca->is_supply_fan_active = new BinarySensor();
    ca->is_supply_fan_active->set_name("Supply Fan");
    App.register_binary_sensor(ca->is_supply_fan_active);
    ca->is_filter_full = new BinarySensor();
    ca->is_filter_full->set_name("Filter Full");
    App.register_binary_sensor(ca->is_filter_full);
    ca->is_summer_mode = new BinarySensor();
    ca->is_summer_mode->set_name("Summer Mode");
    App.register_binary_sensor(ca->is_summer_mode);
    ca->return_air_level = new Sensor("Return Air Level");
    App.register_sensor(ca->return_air_level);
    ca->supply_air_level = new Sensor("Supply Air Level");
    App.register_sensor(ca->supply_air_level);
    App.register_climate(ca);
    return {ca};
    
  climates:
    - name: "wernig"

button:
  - platform: template
    name: "Reset Filter"
    on_press:
      then:
        - lambda: |-
                auto reset_filter = new esphome::comfoair::ComfoAirComponent(id(uart_bus));
                reset_filter->reset_filter();
  - platform: restart
    name: "Wernig ESP Restart"
```

commit 6b007df6 (2024-04-03): Renamed function to reset_and_self_test as this has nothing todo with filter reset. There does not appear to be a filter reset. Closes #18.
closed 2024-04-03 by wichers