wichers/esphome-comfoair #29: Implementing more registers

svenar-nl, 2024-03-30:
Hi,
Currently I'm working on making changes to your code to integrate additional registers.
These include quite some modifications to your code-base.
Are you open to discussing these changes and potentially accepting pull requests related to them?

Here's a brief overview of the changes I'm implementing:
* Moved `messages.h` to `register.h` and added documentation to each item, based on the protocol description.
* Removed `is_` from binary sensors and instead using a list in `__init__.py` to keep track of binary sensors.
* Added text sensors for type, size, filter, preheating valve and frost protection level.
* Implemented more registers.
* Changed WiFi credentials to AP mode and added captive portal

These changes are currently being tested
[image]

My idea is to open pull request for these changes to maintain a single, updated repository for Comfoair and ESPHome.
I'm open for any feedback or suggestions you may have.

Kind regards,
Met vriendelijke groet,
Sven

wichers, 2024-03-30:
Hi Sven,

Sounds good, looking forward in seeing your pull request.

Kind regards,
Alexander

svenar-nl, 2024-03-30:
Alright, glad to hear :)
I'll start by cleaning up my code and putting it into branches for a few PR's instead of one big PR.

wichers, 2024-03-31:
Thank you so much for this addition, that was quite a lot of work from your side! I modified the card to match the new variable names. Any additions for the lovelace card are welcome as well ;)

closed 2024-03-31 by wichers