wichers/esphome-comfoair #35: No sensor data WHR 930 since new version

zoeperman, 2024-04-17:
Hi,

since it updated to the new version (and adjusting the config with sensors)
all the sensors have state unknown in Home assistant

originaly I had this and that worked, I can't figureout how to debug or see what is wrong:

```
external_components:
  - source:
      type: git
      url: https://github.com/wichers/esphome-comfoair
    components: [comfoair]

uart:
  id: uart_bus
  rx_pin: 3
  tx_pin: 1
  baud_rate: 9600

comfoair:
  name: "WHR 930"
  uart_id: uart_bus
  fan_supply_air_percentage:
    name: "Aanvoer ventilator (%)"
  fan_exhaust_air_percentage:
    name: "Afvoer ventilator (%)"
  fan_speed_supply:
    name: "Aanvoer ventilator snelheid"
  fan_speed_exhaust:
    name: "Afvoer ventilator snelheid"
  is_bypass_valve_open:
    name: "Is bypass open?"
  is_preheating:
    name: "Is voorwerwarmen actief?"
  outside_air_temperature:
    name: "Buiten temperatuur"
  supply_air_temperature:
    name: "Aanvoer temperatuur"
  return_air_temperature:
    name: "Retour temperatuur"
  exhaust_air_temperature:
    name: "Afvoer temperatuur"
  return_air_level:
    name: "Retour level"
  supply_air_level:
    name: "Aanvoer level"
  is_supply_fan_active:
    name: "Is toevoer ventilator actief?"
  is_filter_full:
    name: "Filters vol?"
  bypass_factor:
    name: "Bypass factor"
  bypass_step:
    name: "Bypass stap"
  bypass_correction:
    name: "Bypass correctie"
  is_summer_mode:
    name: "Zomerstand?"
```

wichers, 2024-04-17:
The names of your variables are used as labels, this is not recommended. There should be variables within HA that match the name as you've supplied though?

wichers, 2024-04-18:
fyi I had a similar naming scheme but was unable to add a proper prefix to the variables. And giving it more thought, a locale label should be added in the card not in addon.

zoeperman, 2024-04-18:
Using the naming scheme with sensors from the readme.md, makes no difference
powered off the esp didn't help either
[image]

zoeperman, 2024-04-18:
``` yaml
esphome:
  name: esphome-web-f52f1c
  friendly_name: ESP_WTW
  platform: ESP8266
  board: esp01_1m

# Disable uart logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "********"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome_wtw"
    password: "*******"

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/wichers/esphome-comfoair
    components: [comfoair]

uart:
  id: uart_bus
  rx_pin: 3
  tx_pin: 1
  baud_rate: 9600

comfoair:
  name: "WHR 930"
  uart_id: uart_bus
  type:
    name: "type"
  size:
    name: "size"
  intake_fan_speed:
    name: "intake_fan_speed"
  exhaust_fan_speed:
    name: "exhaust_fan_speed"
  intake_fan_speed_rpm:
    name: "intake_fan_speed_rpm"
  exhaust_fan_speed_rpm:
    name: "exhaust_fan_speed_rpm"
  ventilation_level:
    name: "ventilation_level"
  bypass_present:
    name: "bypass_present"
  bypass_valve:
    name: "bypass_valve"
  bypass_valve_open:
    name: "bypass_valve_open"
  preheating_state:
    name: "preheating_state"
  outside_air_temperature:
    name: "outside_air_temperature"
  supply_air_temperature:
    name: "supply_air_temperature"
  return_air_temperature:
    name: "return_air_temperature"
  exhaust_air_temperature:
    name: "exhaust_air_temperature"
  enthalpy_temperature:
    name: "enthalpy_temperature"
  ewt_temperature:
    name: "ewt_temperature"
  reheating_temperature:
    name: "reheating_temperature"
  kitchen_hood_temperature:
    name: "kitchen_hood_temperature"
  return_air_level:
    name: "return_air_level"
  supply_air_level:
    name: "supply_air_level"
  supply_fan_active:
    name: "supply_fan_active"
  filter_status:
    name: "filter_status"
  enthalpy_present:
    name: "enthalpy_present"
  ewt_present:
    name: "ewt_present"
  options_present:
    name: "options_present"
  preheating_present:
    name: "preheating_present"
  bypass_factor:
    name: "bypass_factor"
  bypass_step:
    name: "bypass_step"
  bypass_correction:
    name: "bypass_correction"
  bypass_open_hours:
    name: "bypass_open_hours"
  preheating_hours:
    name: "preheating_hours"
  level0_hours:
    name: "level0_hours"
  level1_hours:
    name: "level1_hours"
  level2_hours:
    name: "level2_hours"
  level3_hours:
    name: "level3_hours"
  preheating_valve:
    name: "preheating_valve"
  frost_protection_active:
    name: "frost_protection_active"
  frost_protection_hours:
    name: "frost_protection_hours"
  frost_protection_minutes:
    name: "frost_protection_minutes"
  frost_protection_level:
    name: "frost_protection_level"
  filter_hours:
    name: "filter_hours"
  motor_current_bypass:
    name: "motor_current_bypass"
  motor_current_preheating:
    name: "motor_current_preheating"
  summer_mode:
    name: "summer_mode"
  p10_active:
    name: "p10_active"
  p11_active:
    name: "p11_active"
  p12_active:
    name: "p12_active"
  p13_active:
    name: "p13_active"
  p14_active:
    name: "p14_active"
  p15_active:
    name: "p15_active"
  p16_active:
    name: "p16_active"
  p17_active:
    name: "p17_active"
  p18_active:
    name: "p18_active"
  p19_active:
    name: "p19_active"
  p90_active:
    name: "p90_active"
  p91_active:
    name: "p91_active"
  p92_active:
    name: "p92_active"
  p93_active:
    name: "p93_active"
  p94_active:
    name: "p94_active"
  p95_active:
    name: "p95_active"
  p96_active:
    name: "p96_active"
  p97_active:
    name: "p97_active"
  bathroom_switch_on_delay_minutes:
    name: "bathroom_switch_on_delay_minutes"
  bathroom_switch_off_delay_minutes:
    name: "bathroom_switch_off_delay_minutes"
  l1_switch_off_delay_minutes:
    name: "l1_switch_off_delay_minutes"
  boost_ventilation_minutes:
    name: "boost_ventilation_minutes"
  filter_warning_weeks:
    name: "filter_warning_weeks"
  rf_high_time_short_minutes:
    name: "rf_high_time_short_minutes"
  rf_high_time_long_minutes:
    name: "rf_high_time_long_minutes"
  extractor_hood_switch_off_delay_minutes:
    name: "extractor_hood_switch_off_delay_minutes"

sensor:
  - platform: uptime
    name: "WHR930 sensor Uptime"
```
[logs_esphome-web-f52f1c_logs.txt](https://github.com/wichers/esphome-comfoair/files/15021267/logs_esphome-web-f52f1c_logs.txt)

wichers, 2024-04-21:
Your log doesn't look right:, there are no sensors being set. Here is a sample from my log:

<pre>
[21:13:42][D][sensor:093]: 'supply_air_temperature': Sending state 13.00000 °C with 1 decimals of accuracy
[21:13:42][D][sensor:093]: 'return_air_temperature': Sending state 19.50000 °C with 1 decimals of accuracy
[21:13:42][D][sensor:093]: 'exhaust_air_temperature': Sending state 10.00000 °C with 1 decimals of accuracy
[21:13:42][D][comfoair:750]: getting error status
[21:13:42][D][text_sensor:064]: 'filter_status': Sending state 'Full'
[21:13:43][D][comfoair:760]: getting bypass control
[21:13:43][D][sensor:093]: 'bypass_factor': Sending state 0.00000  with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'bypass_step': Sending state 0.00000  with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'bypass_correction': Sending state 0.00000  with 0 decimals of accuracy
[21:13:43][D][comfoair:793]: getting operation hours
[21:13:43][D][sensor:093]: 'level0_hours': Sending state 86.00000 h with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'level1_hours': Sending state 117.00000 h with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'level2_hours': Sending state 39.00000 h with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'level3_hours': Sending state 71.00000 h with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'frost_protection_hours': Sending state 99.00000 h with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'bypass_open_hours': Sending state 208.00000 h with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'preheating_hours': Sending state 1.00000 h with 0 decimals of accuracy
[21:13:43][D][sensor:093]: 'filter_hours': Sending state 217.00000 h with 0 decimals of accuracy
[21:13:44][D][comfoair:798]: getting preheating status
[21:13:44][D][text_sensor:064]: 'preheating_valve': Sending state 'Closed'
[21:13:44][D][sensor:093]: 'frost_protection_minutes': Sending state 0.00000 min with 0 decimals of accuracy
[21:13:44][D][text_sensor:064]: 'frost_protection_level': Sending state 'Unknown'
[21:13:45][D][comfoair:803]: getting time delay
[21:13:45][D][comfoair:734]: getting fan status
[21:13:45][D][sensor:093]: 'intake_fan_speed': Sending state 40.00000 % with 0 decimals of accuracy
[21:13:45][D][sensor:093]: 'exhaust_fan_speed': Sending state 40.00000 % with 0 decimals of accuracy
[21:13:45][D][sensor:093]: 'intake_fan_speed_rpm': Sending state 1293.10339 RPM with 0 decimals of accuracy
[21:13:45][D][sensor:093]: 'exhaust_fan_speed_rpm': Sending state 1309.35754 RPM with 0 decimals of accuracy
[21:13:46][D][comfoair:743]: getting valve status
[21:13:46][D][sensor:093]: 'bypass_valve': Sending state 0.00000 % with 0 decimals of accuracy
[21:13:46][D][sensor:093]: 'motor_current_bypass': Sending state 0.00000 A with 1 decimals of accuracy
[21:13:46][D][sensor:093]: 'motor_current_preheating': Sending state 0.00000 A with 1 decimals of accuracy
[21:13:46][D][comfoair:777]: getting sensor data
[21:13:46][D][sensor:093]: 'enthalpy_temperature': Sending state -20.00000 °C with 1 decimals of accuracy
[21:13:47][D][comfoair:783]: getting ventilation level
[21:13:47][D][comfoair:406]: Level 02
[21:13:47][D][sensor:093]: 'return_air_level': Sending state 40.00000 m³ with 1 decimals of accuracy
[21:13:47][D][sensor:093]: 'supply_air_level': Sending state 40.00000 m³ with 1 decimals of accuracy
[21:13:47][D][sensor:093]: 'ventilation_level': Sending state 1.00000  with 0 decimals of accuracy
[21:13:47][D][climate:396]: '' - Sending state:
[21:13:47][D][climate:399]:   Mode: FAN_ONLY
[21:13:47][D][climate:404]:   Fan Mode: LOW
[21:13:47][D][climate:419]:   Current Temperature: 19.50°C
[21:13:47][D][climate:425]:   Target Temperature: 21.00°C
[21:13:48][D][comfoair:788]: getting temperatures
[21:13:48][D][climate:396]: '' - Sending state:
[21:13:48][D][climate:399]:   Mode: FAN_ONLY
[21:13:48][D][climate:404]:   Fan Mode: LOW
[21:13:48][D][climate:419]:   Current Temperature: 19.50°C
[21:13:48][D][climate:425]:   Target Temperature: 21.00°C
[21:13:48][D][sensor:093]: 'outside_air_temperature': Sending state 9.00000 °C with 1 decimals of accuracy
[21:13:48][D][sensor:093]: 'supply_air_temperature': Sending state 13.00000 °C with 1 decimals of accuracy
[21:13:48][D][sensor:093]: 'return_air_temperature': Sending state 19.50000 °C with 1 decimals of accuracy
[21:13:48][D][sensor:093]: 'exhaust_air_temperature': Sending state 10.00000 °C with 1 decimals of accuracy
[21:13:48][D][comfoair:750]: getting error status
[21:13:48][D][text_sensor:064]: 'filter_status': Sending state 'Full'
[21:13:49][D][comfoair:760]: getting bypass control
[21:13:49][D][sensor:093]: 'bypass_factor': Sending state 0.00000  with 0 decimals of accuracy
[21:13:49][D][sensor:093]: 'bypass_step': Sending state 0.00000  with 0 decimals of accuracy
[21:13:49][D][sensor:093]: 'bypass_correction': Sending state 0.00000  with 0 decimals of accuracy
[21:13:49][D][comfoair:793]: getting operation hours
[21:13:49][D][sensor:093]: 'level0_hours': Sending state 86.00000 h with 0 decimals of accuracy
[21:13:49][D][sensor:093]: 'level1_hours': Sending state 117.00000 h with 0 decimals of accuracy
[21:13:49][D][sensor:093]: 'level2_hours': Sending state 39.00000 h with 0 decimals of accuracy
[21:13:49][D][sensor:093]: 'level3_hours': Sending state 71.00000 h with 0 decimals of accuracy
[21:13:49][D][sensor:093]: 'frost_protection_hours': Sending state 99.00000 h with 0 decimals of accuracy
[... cut ...]
  preheating_valve:
    name: "preheating_valve"
  frost_protection_active:
    name: "frost_protection_active"
  frost_protection_hours:
    name: "frost_protection_hours"
  frost_protection_minutes:
    name: "frost_protection_minutes"
  frost_protection_level:
    name: "frost_protection_level"
  filter_hours:
    name: "filter_hours"
  motor_current_bypass:
    name: "motor_current_bypass"
  motor_current_preheating:
    name: "motor_current_preheating"
  summer_mode:
    name: "summer_mode"
```

zoeperman, 2024-04-25:
after trying a different esp and different settings, tried the orifiginal one angain and still nothing worked.
in the end i powered off the wtw unit, turned it back on againg and then the data stream resumed again.

closed 2024-04-25 by zoeperman