wichers/esphome-comfoair #38: Filter Reset

override80, 2024-06-04:
Hi @wichers, i was reading the discussion [here](https://github.com/wichers/esphome-comfoair/issues/12) with @adorobis, and I'm trying to implement filter reset ina CA350. I0m having the same issue. Could this work?

```
static const uint8_t CMD_FILTER_RESET = 0x37;
...
void filter_reset(void) {
    uint8_t reset_cmd[6] = {82, 0, 0, 0, 0, 0};
    this->write_command_(CMD_FILTER_RESET, reset_cmd, sizeof(reset_cmd));
} 
```

I've created a lambda for the reset button like this in the end of comfoair.yaml:

```
button:
  - platform: template
    name: "Reset Filter"
    on_press:
      then:
        - lambda: |-
                auto filter_reset = new esphome::comfoair::ComfoAirComponent();
                filter_reset->filter_reset();
```

but probably this is not even the way to go, because the filter hours does not even reset in my case (instead, it reboots the board).

tieskuh, 2024-11-12:
I managed to get the filter reset working. Please check my fork: https://github.com/tieskuh/esphome-comfoair

override80, 2024-11-27:
> I managed to get the filter reset working. Please check my fork: https://github.com/tieskuh/esphome-comfoair

Hi @tieskuh , how do i get it working forking this repo? I added your reset_filter() function, but ESPHome complains:

[image]

tieskuh, 2024-11-27:
I haven't tried. But you can use my fork if you want.

override80, 2024-11-28:
> I haven't tried. But you can use my fork if you want.

But I have a CA350, not a WHR 950

tieskuh, 2024-11-28:
Might work.. isn’t tested. It is based on the same code so good change it does.

override80, 2024-12-02:
@tieskuh I cannot get it to compile:

```
esphome:
  name: comfoair
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: REDACTED
    gateway: REDACTED
    subnet: 255.255.255.0

# Disable uart logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: REDACTED
ota:
 platform: esphome

sensor:
binary_sensor:

external_components:
  - source:
      type: git
      url: https://github.com/tieskuh/esphome-comfoair
    components: [comfoair]
    refresh: 0s

uart:
  id: uart_bus
  baud_rate: 9600
  tx_pin: TX
  rx_pin: RX

comfoair:
  name: "WHR 950"
  uart_id: uart_bus
  fan_supply_air_percentage:
    name: "Supply fan"
  fan_exhaust_air_percentage:
    name: "Exhaust fan"
  fan_speed_supply:
    name: "Supply fan speed"
  fan_speed_exhaust:
    name: "Exhaust fan speed"
  is_bypass_valve_open:
    name: "Bypass"
  is_preheating:
    name: "Preheating"
  outside_air_temperature:
    name: "Outside temperature"
  supply_air_temperature:
    name: "Supply temperature"
  return_air_temperature:
    name: "Return temperature"
  exhaust_air_temperature:
    name: "Exhaust temperature"
  is_supply_fan_active:
    name: "Supply fan"
  is_filter_full:
    name: "Filter status"
  bypass_step:
    name: "Bypass valve"
  is_summer_mode:
    name: "Summer mode"

button:
  - platform: template
    name: "Reset filter"
    on_press:
      then:
        - lambda: |-
              id(comfoair_climate)->reset_filter();
```

It sais:

```
INFO ESPHome 2024.11.2
INFO Reading configuration /config/esphome/comfoair.yaml...
INFO Updating https://github.com/tieskuh/esphome-comfoair@None
Failed config

button.template: [source /config/esphome/comfoair.yaml:77]
  platform: template
  name: Reset filter
  on_press: 
    - then: 
        - 
          Couldn't find ID 'comfoair_climate'. Please check you have defined an ID with that name in your configuration.
          lambda: !lambda |-
            id(comfoair_climate)->reset_filter();
  disabled_by_default: False
```

tieskuh, 2024-12-02:
Thats weird. I have no problem compiling it. Just different hardware, the rest of my config is the same.

override80, 2024-12-04:
For whoever lands here: thanks to @tieskuh input, the reset also works with @wichers repo. Just add the "id: comfoair" in the comfoair section:

```
comfoair:
  name: "ComfoAir 350"
  id: "comfoair_climate"
  uart_id: uart_bus
  type:
    name: "comfoair_type"
  size:
    name: "comfoair_size"
  intake_fan_speed:
    name: "comfoair_intake_fan_speed"
  exhaust_fan_speed:
    name: "comfoair_exhaust_fan_speed"
...
```

and then at the end add the button like this:

```
button:
  - platform: template
    name: "Reset Filter"
    on_press:
      then:
        - lambda: |-
                id(comfoair_climate)->filter_reset();
```

closed 2024-12-04 by override80