wichers/esphome-comfoair #40: Fan Speed Incorrectly Displayed & Unable to Control Fan Speed

roberto2lini, 2024-06-11:
Firmware: 2024.5.5 (Jun 7 2024, 13:39:41)
HA version: 2024.6.1 & 2024.6.2

I have an WHR 930R, and for some reason the fan speed is not measured / displayed correctly on the WebGUI or in HA. I have a 3 speed selector for speeds 1,2,3. If I set the speed to 1, the WHR930 screen also displays 1. But the ESPHome WebGUI and HA reports the fan to be off, setting the fan speed to 2, the WHR930 screen also displays 2 and the ESPHome WebGUI and HA report "Low". 

From the screenshots you can see the ComfoAir Returns Level 2 but the Climate Log line reports as Low. 
And for Level 1 -  ComfoAir Returns Level 1 but the Climate Log line reports as Off. 

Maybe this is the reason for which I cant set the fan speed via the ComfoAir Controls. It keeps jumping back to the currently set option. 

[image]
[image]

tolgaeren, 2026-09-20:
> Can you let me know if this is still an issue?

For me yes, my WHR930 behaves the same way as described above.

tolgaeren, 2026-09-21:
What the firmware version of your WHR @roberto2lini ?
my is at `WHR 930 v3.01 b32`

bruvv, 2026-10-25:
I have done some updating can this be closed or can we get an update @roberto2lini and @tolgaeren

bruvv, 2026-11-14:
closing this because of a lag of reply

closed 2026-11-14 by bruvv