wichers/esphome-comfoair #41: Auto and Off mode do not work

svenar-nl, 2024-07-10:
When selecting fan mode 'Auto' or 'Off' it shows the following message `Failed to call service climate/set_hvac_mode. expected HVACMode or one of 'off', 'heat', 'cool', 'heat_cool', 'auto', 'dry', 'fan_only' for dictionary value @ data['hvac_mode']`.

bfabio, 2024-08-14:
Same here (Comfoair 160)

deam0n, 2024-09-04:
I guess that is normal... from the confoair.h only fan_mode is exposed:

`traits.set_supported_modes({
      climate::CLIMATE_MODE_FAN_ONLY
    });`
    
 So if you want to change the fans speed you'll have to use the action: climate.set_fan_mode and not climate.set_hvac_mode

bfabio, 2024-09-05:
@deam0n wondering if it's the right abstraction or if there's something we can do to make it work out of the box (I'm not an expert on ESPHome)

wichers, 2025-08-01:
This is expected behavior if you're using hvac functions.

closed 2025-08-01 by wichers