wichers/esphome-comfoair #50: Works with ComfoAir 160

bfabio, 2024-08-09:
Hi, this is not actually an issue, just some info: it works perfectly with my ComfoAir 160 unit, thanks for the project @wichers!

wichers, 2024-08-09:
> Maybe we can add a section in the README with all known to be compatible models.
Sounds like a good idea.

zanac, 2025-07-31:
Hi, where you connected to comfoair 160? Did you use RJ45, could you give me some info?

ErichGr, 2025-08-01:
I would recomment to put some resistors in the Tx lines (maybe 1k or 100 Ohms) because on my System the RS232 are not a real RS232 signals in terms of voltages on the lines. Without the resistors the your MAX 232 draws toooo much current, will get very hot and will break soon.
If your signals have proper RS232 voltages ignore my comment otherwise put the resistors.
BR

wichers, 2025-08-01:
@ErichGr what type of unit is this for? Sounds like this is rs485 instead of rs232.

zanac, 2025-08-01:
> [@zanac](https://github.com/zanac) I used the RS232 port with a [MAX232](https://en.wikipedia.org/wiki/MAX232) module. Give me a week and I'll post the photos.

could you contact me via private way for just a few question about cable and connection? i already made CA350 connection and now i want to help a my friend that own a Comfoair 160.... you can contact me with facebook (my name is "Vanni Brutto"). I really thank you if you can...

wichers, 2025-08-01:
Sorry I don't have facebook, you could write your email here.

bfabio, 2025-08-26:
@zanac here's my setup. Note that the wires change colors, because different cables have been spliced together, but you can easily get the idea.

[image]

[image]

[image]