wichers/esphome-comfoair #54: Partially working in combination with ComfoSense

LeoSum8, 2024-08-30:
Hi there,
first of all thank you @wichers for this great component.
It is working perfectly if I connect the ESP to my Comfoair 180.

However, as a next step, I wanted to also hook up the ComfoSense again, to still allow manual control.

So I attached it to another UART via a second RS232-TTL converter and added the uart_mitm (https://github.com/ssieb/esphome_components/tree/master/components/uart_mitm) as another custom component.

As a result, I can now control the Comfoair both via Home Assistant and manually, but reading its status doesn't work anymore. Neither on the Comfosense, nor in Home Assistant (to be precise, the Comfosense still displays fan speed (away, low, mid, high) but only if it was chaned via Comfosense. When Changed via Home Assistant, the fan physically changes, but the change is not displayed in the Comfosense).

It seems as if the TX line from the Comfosense to the ESP was interrupted, but I think that is not the case, because the Comfosense only reports a comm-error, when I disconnect the communication between comfosense and ESP. As long as that is still attached, the comfosense behaves normally.

Any ideas? Is anyone runing a similar setup?

Here is my config:

```
esphome:
  name: comfoair

esp8266:
  board: nodemcu

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "redacted"

ota:
  - platform: esphome
    password: "redacted"

wifi:
  ssid: "redacted"
  password: "redacted!"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ComfoAir Fallback Hotspot"
    password: "redacted"

captive_portal:
    
external_components:
  - source:
      type: git
      url: https://github.com/wichers/esphome-comfoair
    components: [comfoair]
  - source:
      type: git
      url: https://github.com/ssieb/esphome_components
    components: [uart_mitm]

uart:
  - id: uart_comfoair
    rx_pin: GPIO13
    tx_pin: GPIO15
    baud_rate: 9600
  - id: uart_comfosense
    rx_pin: GPIO12
    tx_pin: GPIO14
    baud_rate: 9600

uart_mitm:
  uart1: uart_comfoair
  uart2: uart_comfosense

comfoair:
  name: "ComfoAir 180"
  uart_id: uart_comfoair
  type:
    name: "type"
  size:
    name: "size"
  intake_fan_speed:
    name: "intake_fan_speed"
  exhaust_fan_speed:
    name: "exhaust_fan_speed"
  intake_fan_speed_rpm:
    name: "intake_fan_speed_rpm"
  exhaust_fan_speed_rpm:
    name: "exhaust_fan_speed_rpm"
  ventilation_level:
    name: "ventilation_level"
  bypass_present:
    name: "bypass_present"
  bypass_valve:
    name: "bypass_valve"
  bypass_valve_open:
    name: "bypass_valve_open"
  preheating_state:
    name: "preheating_state"
  outside_air_temperature:
    name: "outside_air_temperature"
  supply_air_temperature:
    name: "supply_air_temperature"
  return_air_temperature:
    name: "return_air_temperature"
  exhaust_air_temperature:
    name: "exhaust_air_temperature"
  enthalpy_temperature:
    name: "enthalpy_temperature"
  ewt_temperature:
    name: "ewt_temperature"
  reheating_temperature:
    name: "reheating_temperature"
  kitchen_hood_temperature:
    name: "kitchen_hood_temperature"
  return_air_level:
    name: "return_air_level"
  supply_air_level:
    name: "supply_air_level"
  supply_fan_active:
    name: "supply_fan_active"
  filter_status:
    name: "filter_status"
  enthalpy_present:
    name: "enthalpy_present"
  ewt_present:
    name: "ewt_present"
  options_present:
    name: "options_present"
  preheating_present:
    name: "preheating_present"
  bypass_factor:
    name: "bypass_factor"
  bypass_step:
    name: "bypass_step"
  bypass_correction:
    name: "bypass_correction"
  bypass_open_hours:
    name: "bypass_open_hours"
  preheating_hours:
    name: "preheating_hours"
  level0_hours:
    name: "level0_hours"
  level1_hours:
    name: "level1_hours"
  level2_hours:
    name: "level2_hours"
  level3_hours:
    name: "level3_hours"
  preheating_valve:
    name: "preheating_valve"
  frost_protection_active:
    name: "frost_protection_active"
  frost_protection_hours:
    name: "frost_protection_hours"
  frost_protection_minutes:
    name: "frost_protection_minutes"
  frost_protection_level:
    name: "frost_protection_level"
  filter_hours:
    name: "filter_hours"
  motor_current_bypass:
    name: "motor_current_bypass"
  motor_current_preheating:
    name: "motor_current_preheating"
  summer_mode:
    name: "summer_mode"
  p10_active:
    name: "p10_active"
  p11_active:
    name: "p11_active"
  p12_active:
    name: "p12_active"
  p13_active:
    name: "p13_active"
  p14_active:
    name: "p14_active"
  p15_active:
    name: "p15_active"
  p16_active:
    name: "p16_active"
  p17_active:
    name: "p17_active"
  p18_active:
    name: "p18_active"
  p19_active:
    name: "p19_active"
  p90_active:
    name: "p90_active"
  p91_active:
    name: "p91_active"
  p92_active:
    name: "p92_active"
  p93_active:
    name: "p93_active"
  p94_active:
    name: "p94_active"
  p95_active:
    name: "p95_active"
  p96_active:
    name: "p96_active"
  p97_active:
    name: "p97_active"
  bathroom_switch_on_delay_minutes:
    name: "bathroom_switch_on_delay_minutes"
  bathroom_switch_off_delay_minutes:
    name: "bathroom_switch_off_delay_minutes"
  l1_switch_off_delay_minutes:
    name: "l1_switch_off_delay_minutes"
  boost_ventilation_minutes:
    name: "boost_ventilation_minutes"
  filter_warning_weeks:
    name: "filter_warning_weeks"
  rf_high_time_short_minutes:
    name: "rf_high_time_short_minutes"
  rf_high_time_long_minutes:
    name: "rf_high_time_long_minutes"
  extractor_hood_switch_off_delay_minutes:
    name: "extractor_hood_switch_off_delay_minutes"

i2c:
  sda: GPIO4
  scl: GPIO0

bme680_bsec:
  address: 0x77
  temperature_offset: 0

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    iaq:
      name: "BME680 IAQ"
      id: iaq
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      name: "BME680 IAQ Accuracy"

  - platform: template
    name: "BME680 IAQ Classification"
    icon: "mdi:checkbox-marked-circle-outline"
    lambda: |-
      if ( int(id(iaq).state) <= 50) {
        return {"Excellent"};
      }
      else if (int(id(iaq).state) >= 51 && int(id(iaq).state) <= 100) {
        return {"Good"};
      }
      else if (int(id(iaq).state) >= 101 && int(id(iaq).state) <= 150) {
        return {"Lightly polluted"};
      }
      else if (int(id(iaq).state) >= 151 && int(id(iaq).state) <= 200) {
        return {"Moderately polluted"};
      }
      else if (int(id(iaq).state) >= 201 && int(id(iaq).state) <= 250) {
        return {"Heavily polluted"};
      }
      else if (int(id(iaq).state) >= 251 && int(id(iaq).state) <= 350) {
        return {"Severely polluted"};
      }
      else if (int(id(iaq).state) >= 351) {
        return {"Extremely polluted"};
      }
      else {
        return {"error"};
      }
```

LeoSum8, 2024-08-30:
I was a little too impatient. The Home Assistant readings do update sporadically. But in the previous setup without the comfosense, it was giving readings more often and updates were a lot quicker.

Is the mitm component somehow conflicting with this component in a way that it clears the buffer and data that is read by mitm is then no longer available to esphome-comfoair?
Is this a race-condition problem?

bruvv, 2026-10-30:
@LeoSum8 can you retry with the newest version?

LeoSum8, 2026-11-13:
Hi @bruvv , thanks for picking this up!
I can find some time to test this next week.
How do I wire it up to test with the latest commit? Do I daisy chain the comfosense through my esp or do I connect the two in parallel?

bruvv, 2026-11-14:
Hi @LeoSum8, thanks for testing!

You should connect the ComfoSense **through the ESP**, not in parallel:

`ComfoSense → RS232 adapter → ESP → RS232 adapter → ComfoAir`

Use two separate UARTs and replace `uart_mitm` with the built-in `proxy_uart`:

```yaml
uart:
  - id: uart_comfoair
    rx_pin: GPIO13
    tx_pin: GPIO15
    baud_rate: 9600

  - id: uart_comfosense
    rx_pin: GPIO12
    tx_pin: GPIO14
    baud_rate: 9600

comfoair:
  name: "ComfoAir 180"
  uart_id: uart_comfoair
  proxy_uart: uart_comfosense
```

Use the latest `master` commit for testing.

The proxy has software test coverage, but hasn't been verified with a physical ComfoSense yet. It would be great if you could check whether both the ComfoSense and Home Assistant can change the fan speed, whether status updates work in both directions, and whether the connection remains stable.

Please share your ESPHome logs and findings!

papadi, 2026-11-16:
You were having this conversation right when I was about to try all this. I did and it works fine. Comfosense sees the ventilation level changes made in Home Assistance and vice-versa.

The only side effect I noticed is that when level changes in Home Assistance, the "Manual" indication starts blinking on Comfosense which is the indication that it is in temporary manual mode and the next scheduled even in Comfosense will take over. So, Home Assistance changes are considered by Comfosense as manual changes, which I guess makes sense. Anyway, not sure I want to have schedules configured in my Comfosense while I have the whole thing set up wit Home Assistant. The only purpose of ComfoSense is this set up is backup, if something goes wrong with my home assistant.