wichers/esphome-comfoair #56: Reads stopped working with G90 380 - question

scarfuzz, 2025-01-01:
Hi wichers,
First of all a huge thank you for this implementation. It worked successfully for a long while but at some point after updating HA and/or this implementation the status updates stopped working. I still can successfully do writes and control the fan speed. The read response is indicated with the flashing light on the JY-R2T which follows the polling of the get-functions. I replicated the behavior on a second ESP8266 still the sensors remain undefined in HA. Before I also swap the JY-R2T and/or fork the repo to do some debug prints to check if there are any values fetched: Have you ever experienced something similar? 
Thanks a lot
scarfuzz

rom4ikk, 2025-03-28:
Hey, I see the same thing. My integration stops working after a time. 

Did you found the problem? Or is there any solution? 

I already replace the esp32 and the rs232 max. But still the same problem. It's working for a time and then just everything stops(undefined) 

Best regards 
Roman

scarfuzz, 2025-08-01:
Hi guys,
Sorry for not being really active these days. For me reads stopped working at all. For now I removed my installation and I'm back to use the CC Ease. I got a bit tired of the debugging. Not sure if I will give it another try in the near future. Hope you are able to resolve the issues. Thanks!
Wolfgang