wichers/esphome-comfoair #57: Not compatible with Zehnder WHR918?

crysxd, 2025-01-31:
Just for other people to find this: 
I think this is not compatible with the WHR 918. Many resellers list the unit as "WHR 918 / ComfoAir 180" which lead me to believe it's the same unit but with a market specific model name. There is one subtle difference though in the data sheets: The ComfoAir 180 lists the RS232 interface for home automation, the WHR 918 does not. I also found this [page](https://zehndernl.zendesk.com/hc/nl/articles/360002694497-WHR-uitlezen-via-Domoticz) which seems to confirm this. Nevertheless the WHR 918 does have the green connector with GND, RX, TX, 12V on the front and comes even with a little screw terminal plug to use the port and the electrical schema provided by Zehnder shows this green connector being internally wired to the RS232 port (same as with the ComfoAir 180)

| Schematic| Green connector on front |
|---|---|
|[image]|[image]|

I tried it nevertheless but I get no response back from the machine, with debugging enabled on the UART port I get following logs:
```
[15:17:34][D][comfoair:478]: getting error status
[15:17:34][D][uart_debug:158]: >>> "\a\xF0\x00\xD9\x00\x86\a\x0F"
[15:17:35][D][comfoair:488]: getting bypass control
[15:17:35][D][uart_debug:158]: >>> "\a\xF0\x00\xDF\x00\x8C\a\x0F"
[15:17:36][D][comfoair:463]: getting fan status
[15:17:36][D][uart_debug:158]: >>> "\a\xF0\x00\v\x00\xB8\a\x0F"
[15:17:37][D][comfoair:471]: getting valve status
[15:17:37][D][uart_debug:158]: >>> "\a\xF0\x00\r\x00\xBA\a\x0F"
[15:17:39][D][comfoair:511]: getting ventilation level
[15:17:39][D][uart_debug:158]: >>> "\a\xF0\x00\xCD\x00z\a\x0F"
[15:17:40][D][comfoair:516]: getting temperatures
[15:17:40][D][uart_debug:158]: >>> "\a\xF0\x00\xD1\x00~\a\x0F"
```

Swapping RX/TX does not seem to help, nothing comes back from the machine. Unless I overlook something it seems like the WHR918 does not respond to requests via the RS232 connector.

This is my setup, giving the above logs and blinking TX LED on the UART/RS232 adapter:

[image]

crysxd, 2025-01-31:
I didn't give up and tried some more...turns out the SP3232 was not working seemingly. I swapped the SP3232 board with a different one and.....it works!

Thank you for this amazing project!

[image]

closed 2025-01-31 by crysxd

VincentSC, 2025-02-09:
Hoi @crysxd, Which RS232->TTL board did you use in the end? The one on the photo I could find on AliExpress as "Single Channel 50x15" - did you use that one again?

Reason I ask: as I don't get any signal with a standard MAX3232 on 3.3V, which I probably baked by trying all possible options of RX and TX, I want to order one which I can trust to work.

crysxd, 2025-02-09:
Here is my full setup

### RS232 -> TTL
https://vi.aliexpress.com/item/1005006210630153.html?spm=a2g0o.order_list.order_list_main.55.7cc61802hg1agg&gatewayAdapt=glo2vnm

[image]

### RS232 -> TTL
This is how I have it wired up:

[image]

wichers, 2025-02-09:
Hi Vincent,

Before getting my first version to work I actually blew up the two rs232
level converter NPN transistors on the mainboard. This made the board
unresponsive, so I ended up replacing those. I hope this is not the case
with you, but you can check with a multimeter if yours are faulty. They are
in a SOT23 package if I recall correctly.

Good luck