wichers/esphome-comfoair #61: Circuit Diagram: Short circuit of the RS232 output to ground

jf-64, 2025-07-23:
Hi,

I am currently working on replicating your excellent ESPHome-ComfoAir project and have been reviewing the hardware schematic in detail.
I've encountered a potential ambiguities regarding the wiring and few points in the BOM.

1. **Issue with U3.14 (T1OUT) connected to the ground line (J1.2 - U2.2 - _U3.15(!)_ - J2.2)**
- In the [Schematic](https://github.com/julianpas/esphome-comfoair/blob/master/doc/esp_comfoair.pdf), `U3.14 (T1OUT)` (the RS232 output of the MAX3232) appears to be directly connected to `U3.15` and with the ground line.
This would result in a **short circuit of the RS232 output to ground**, which could damage the chip or prevent its function.
Is this connection intended, or is it a diagrammatic error? Typically, `T1OUT` would connect to the RS232 RX pin of the ComfoAir.
2. **BOM vs. Circuit Diagram (CD)**
- BOM: MAX 232 DIP RS232, CD: MAX3232
- BOM: 12V zu 5V 3A 15W DC-DC Step Down Buck Converter, CD: 12V zu 3.3V 1.5A 15W DC-DC Step Down Buck Converter
- BOM: C1-C5 are missing

ErichGr, 2025-08-01:
I would recomment to put some resistors in your Tx lines (maybe 1k or 100 Ohms) because on my System the RS232 is not a real RS232 in terms of voltages on the lines. Without the resistors the your MAX 232 draws toooo much current, will get very hot and will break soon.
BR

wichers, 2025-08-01:
@jf-64 The schematic you're referring to is not mine. If you have an updated correct schematic, I'll be glad to add it to the documentation.

@ErichGr like I mentioned in the other issue, yours could be RS485, what type of unit is this?